ironsheep/lightning-detector-MQTT2HA-Daemon #7: Could you add an SPI interface?

terrycarlin, 2020-07-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

The current version (2) of SparkFun Lightning Detector - AS3935 comes with an SPI interface
The SparkFun clams the I2C interface is not reliable.
The I2C interface is still on the board, but looks like you have to cut traces on the board to enable it.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Add a SPI interface to the Daemon

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Cutting traces is a last resort and may cause the board to be unreliable due to the I2C chipset.
I would probably attempt to add the SPI interface.
 
**Additional context**
Add any other context or screenshots about the feature request here.

My board arrives tomorrow (July 20). Looking forward to playing with it in HA.
Thx

ironsheep, 2020-08-01:
I can help if you are willing to do a bit of the work ;-)  The two interfaces from a python code perspective should be pretty similar.

At first blush the worklist appears to be:
- locate SPI python code that we can pattern after (do you have any, or know of, working examples?)
- adapt the current code to select between SPI/i2c with new config.ini setting
- add the SPI code and runtime switch between
- test the new SPI support (code and setup docs) against a sensor board... to prove it all working
- adjust the setup part of the README to describe what's needed to enable SPI (or i2C) depending on which the user wants.
- release the new version to the repo.

I'm in the midst of a number of small projects (pretty much my life all the time ;-)  but I can put some time into this if you will too.

Thoughts?

terrycarlin, 2020-08-01:
Will give it a shot. I have been looking at some python code for RPis that talk to SPI.
Will wire up my Lightning Sensor and poke at it.

Thanks

Terry Carlin

ironsheep, 2020-08-01:
Nice, I'll lead off while you are wiring up by adding the config option in the code to select between SPI/I2C... I'll send a .zip to you... you can contact me at stephen@ironsheep.biz by email and I'll send it by return email...

lightmaster, 2020-08-15:
Any progress with this?

terrycarlin, 2020-08-17:
Hi there,

Not having too much luck. I have written a test program to try the different methods I found to set the registers in the AS3935. All I get back is zeros in every register. I have checked the wiring many times just to make sure it’s hooked up right.

I took a bunch of code from the RaspberryPi-AS3935 library to test out the code.
I found Python libraries that do SPI and C libraries that do SPI but none of the code works with the RPi and lets you update a register. There are plenty of examples of reading/writing to SPI treating it like a sequential file. I tried to set the board into loopback mode so I could see if the board sees what I am sending but the  C++ library called Py-SPIDEV that I am using seems to have a bug and won’t let me set loopback I wrote a note to SparkFun asking them for some help/examples week before last but haven’t heard anything.

I will keep poking at it. I am also open to any advise you have. I was hoping to have this done quickly as here in Arizona it is monsoon season and there is usually plenty of thunder storms. We are having one right now. 

This is how I hooked up the wiring:

SPI

AS3935 Pin	Module Pin	Raspberry Pi Pin
4 (GND)	GND	9 (Ground)
5 (VDD)	3.3V	1 (3.3v)
10 (IRQ)	INT	11 (GPIO 17)
8 (CS )	CS	24 (SPI CS0)
11 (SCL)	CLK	23 (CLK)
12 (MOSO)	MOSO	21 (SPI MOSO)
13 (MOSI)	MOSI	19 (SPI MOSI
I get a red light on the board when I have power on the SPi. I am assuming that is correct.
I never got any smoke or melting components when I powered it on.

lightmaster, 2020-08-17:
I was planning to attempt this by moving HA from the Hassio minimalist OS to full Raspbian running Hassio Supervised in Docker, however my RPi 3B seems to be running out of memory with HA and everything going when it's installed on Raspbian. Memory usage upwards of 800MB then locks up. Can't figure out how to install Python3 on the Hassio OS version yet.

terrycarlin, 2020-08-17:
I thought Hassio already had Python 3 installed. HA won’t run on anything less than 3.7 or so. 

If it ain't broke, take it apart and see how it works.

lightmaster, 2020-08-19:
I've moved HA onto my DS920+ and am gonna setup the RPi 3B to try to figure out the SPI interface of this detector. Goal will be to have the RPi 3B interface over SPI with the detector board, and the RPi will connect to HA's MQTT broker that's running on my DS920+, that way HA and all its addons get to benefit from 20GB of RAM and won't be running out of memory like it was on the RPi 3B.

lightmaster, 2020-08-25:
> I took a bunch of code from the RaspberryPi-AS3935 library to test out the code.

I tried out this library too, but it would not connect, with the lightning.connected variable always reporting as `false` even when using a very simplified code. I ended up using the steps at [SparkFun](https://learn.sparkfun.com/tutorials/sparkfun-as3935-lightning-detector-hookup-guide-v20/all#spi-only) to set the board to I<sup>2</sup>C. So far the strikes nearby that I can hear myself correspond to to an increase on the Lovelace Card.

terrycarlin, 2020-08-25:
> I tried out this library too, but it would not connect, with the lightning.connected variable always reporting as false even when using a very simplified code. I ended up using the steps at SparkFun <https://learn.sparkfun.com/tutorials/sparkfun-as3935-lightning-detector-hookup-guide-v20/all#spi-only> to set the board to I2C. So far the strikes nearby that I can hear myself correspond to to an increase on the Lovelace Card.

I am trying to get SparkFun to show us how to set the registers using Python on a RPi. That is where I got stuck. I was tempted to modify my sensor to use I2C but decided against that as SparkFun says that I2C is not supported by them because it’s not reliable.

Right now, while I am waiting on a response from SparkFun, I have taken a different approach. SparkFun has a AS3935 SPI library for Arduinos. I happen to have a couple hanging around so I loaded that library up. SparkFun also has a couple of demo Sketches that use SPI to talk to the sensor. I loaded up one of those and ran it. It detected the sensor and since I am in Arizona and we are having Monsoon season, there is usually lightning in the air often so it makes it pretty easy and fun to code and test.
I have the Arduino sending a Json packet over the USB line every time the lightning strikes. It also flashes a RGB diode on the Arduino board just for fun. I have written a Python script that reads from the USB line and prints out the strike info.. I was meditating about adding this functionality to a copy ISP-lightning-mqtt-daemon.py. just to see if I can make it work. I am continuing to polish the Arduino Sketch. 
Here is one of the packets:

{“strike”:{"distance": 3.11, "energy": 56848, "unit_of_measurement": "mi”}}

fugazzy, 2020-08-26:
I ended up using the steps at [SparkFun](https://learn.sparkfun.com/tutorials/sparkfun-as3935-lightning-detector-hookup-guide-v20/all#spi-only) to set the board to I2C. So far the strikes nearby that I can hear myself correspond to to an increase on the Lovelace Card.

I am having the same issues, but I could not solve them yet. Here I described what I tried: 
https://community.home-assistant.io/t/lovelace-lightning-detector-card-for-as3935/214167/6?u=fugazzy
Can you give me a hint?

ironsheep, 2020-09-02:
FYI- the new SPI addition just started working today.  I'm finishing the addition an antenna tune portion next.

terrycarlin, 2020-09-02:
Great!

lightmaster, 2020-09-02:
Tomorrow, I'll reconfigure my sensor to use SPI like the manufacturer suggests and see how well the [develop branch](/ironsheep/lightning-detector-MQTT2HA-Daemon/tree/develop) works. Well, suppose I'll wait around until the next storm to actually make sure that it's working, but it is summer here in the Southeast US, so the next storm should be within a few days.

ironsheep, 2020-09-02:
@lightmaster good, I'll look forward to hearing of your results.

ironsheep, 2020-09-02:
@terrycarlin take a look at this: https://github.com/ironsheep/lightning-detector-MQTT2HA-Daemon/blob/develop/README-SPI.md

Please let me know what you think...

lightmaster, 2020-09-02:
When disabling I<sup>2</sup>C on the Sparkfun AS3935, after you cut `CS` and reconnect `SI`, does it matter whether or not you also cut the I<sup>2</sup>C `PU`?

ironsheep, 2020-09-02:
Only cut/short the trace(s)/pad(s) indicated by sparkfun in their docs. 
I would do the least amount of work to get SPI to work. If you want to go back to I2C you'll have less to undo.

Which version of their board do you have? Once I know I can then give a more precise answer if needed...

lightmaster, 2020-09-02:
The same version depicted at [the Sparkfun guide v20](https://learn.sparkfun.com/tutorials/sparkfun-as3935-lightning-detector-hookup-guide-v20). I followed their directions to enable I<sup>2</sup>C originally, so I cut the trace `SI` and connected `CS`, then connected the middle and right pads on the I<sup>2</sup>C `PU`.

lightmaster, 2020-09-02:
Its now setup to use SPI using the [develop branch](https://github.com/ironsheep/lightning-detector-MQTT2HA-Daemon/tree/develop). After starting PiGPIOd before manually running the script, I didn't get any errors. I set the service to log to syslog so I can see any errors that might popup and its now running as a service, waiting for a storm to roll by.

ironsheep, 2020-09-02:
Good to hear!  Looking at your readme adjustments.

Regards,
Stephen
--

Stephen M Moraco
Iron Sheep Productions, LLC

ironsheep, 2020-09-04:
Your pull request is merged and this SPI support is now released and avail in v2.2.0 and later.

closed 2020-09-04 by ironsheep

terrycarlin, 2020-09-04:
Way kewl!