ironsheep/lightning-detector-MQTT2HA-Daemon #8: MQTT lightningdetector/detect not updating count and distance

lightmaster, 2020-08-25:
**Checklist:**

- [x] I updated to the latest version available
- [x] I double-checked my hardware connections
- [x] I checked that my MQTT broker is otherwise working

**Release with the issue:**
v2.0.3
**Last working release (if known):**

**Hardware, Operating System, Python version:**
RPi 3B, Raspberry Pi OS, Python 3.7.3

**Description of problem:**
MQTT messages to `detect` for `count` and `distance` always equal 1, even though the messages sent to `crings` and `prings` count properly and show reasonable distances.

[image]

**Python errors shown in the logs (if applicable):**

```

```

**Additional information:**

ironsheep, 2020-08-27:
A review of code shows nothing notable.  I'm going to have to recreate this to find the problem.  More soonish....

ironsheep, 2020-09-06:
@lightmaster I'm guessing that our latest finding is the explanation for this lack of reporting?  I'll bet when you proved the AS3935 is communicating this will be better!

ironsheep, 2020-09-11:
Is this fixed too then?

lightmaster, 2020-09-11:
I need some lightning to know that, lol. Where's Thor when you need him?

ironsheep, 2020-09-11:
I'll leave this open so you can post status when you finally hear some...  ;-)

lightmaster, 2020-09-13:
Nope, MQTT is still showing 1 count on the Home Assistant MQTT Integrations card.

[image]

[image]

ironsheep, 2020-09-13:
I'm not seeing anything wrong here.  

The count should be the count of detections in the last 3 seconds (it only reports every three seconds so we don't storm the MQTT broker.)  If detections are 3 seconds apart, or more, this will be 1.

re: my alternative to your pull request (*Thank you for it and for noting that I should fix this stuff, as always!*)
Changes committed to /master and appear in [v2.2.7](https://github.com/ironsheep/lightning-detector-MQTT2HA-Daemon/releases/tag/v2.2.7) and later.

ironsheep, 2020-09-17:
Saw your "works great" on my commit.  Good to hear. Closing this issue then.

lightmaster, 2020-09-17:
I was in the middle of typing up something about it on here and gonna close the issue when I got called away at work.

ironsheep, 2020-09-17:
No worries, you can always enter more after closing. But I re-opened it for you.

closed 2021-04-22 by ironsheep