ironsheep/lightning-detector-MQTT2HA-Daemon #11: i2c broken?

Zebble, 2020-09-08:
**Checklist:**

- [X ] I updated to the latest version available
- [ X] I double-checked my hardware connections
- [ X] I checked that my MQTT broker is otherwise working

**Release with the issue:**
master

**Last working release (if known):**
unknown

**Hardware, Operating System, Python version:**
Pi 3B, python 3.7.3, raspbian 10.4

**Description of problem:**

When using i2c, a 'bad i2c bus' error is triggered.

**Python errors shown in the logs (if applicable):**

Traceback (most recent call last):
  File "ISP-lightning-mqtt-daemon.py", line 937, in <module>
    detector = AS3935_I2C(interrupt_pin, i2c_address, i2c_bus)
  File "/opt/ISP-lightning-mqtt-daemon/AS3935/AS3935_i2c_spi.py", line 559, in __init__
    self.device = self.pi.i2c_open(bus, address)
  File "/usr/local/lib/python3.7/dist-packages/pigpio.py", line 2674, in i2c_open
    self.sl, _PI_CMD_I2CO, i2c_bus, i2c_address, 4, extents))
  File "/usr/local/lib/python3.7/dist-packages/pigpio.py", line 1009, in _u2i
    raise error(error_text(v))
pigpio.error: 'bad I2C bus'

```

```

**Additional information:**

It appears the AS3935_I2C call has address and bus reversed.  Changing it to:

    detector = AS3935_I2C(interrupt_pin, i2c_address, i2c_bus)

Fixes this problem, but creates another:

Traceback (most recent call last):
  File "ISP-lightning-mqtt-daemon.py", line 947, in <module>
    detector.set_noise_floor(testValue)
  File "/opt/ISP-lightning-mqtt-daemon/AS3935/AS3935_i2c_spi.py", line 194, in set_noise_floor
    self.write_byte(0x01, (self.read_byte(0x01) & 0b10001111) + ((noise_floor & 0x07) << 4))
  File "/opt/ISP-lightning-mqtt-daemon/AS3935/AS3935_i2c_spi.py", line 576, in read_byte
    return self.pi.i2c_read_byte_data(self.i2c_device, address)
AttributeError: 'AS3935_I2C' object has no attribute 'i2c_device'

Uncommenting line 42 in AS3935_i2c_spi.py fixes the issue:

        # Let our derived classes handle this!
        self.i2c_device = self.pi.i2c_open(bus, address)

Perhaps SPI has become the preferred choice and i2c got broken during the process/addition of adding SPI support?

ironsheep, 2020-09-08:
Ok, no change in preferred. I just broke the code when I reshaped the library into a base and two derived classes.

Changes committed to /master and appear in v2.2.3 and later.

Please advise if this fixes things for you.  (Thanks for reporting this!)

Zebble, 2020-09-08:
Seems to work just fine!  Thanks @ironsheep!

I've ran into some inconsistencies in the docs as well.  I'll do a pull request shortly and help clean them up.

closed 2020-09-08 by Zebble