ironsheep/lightning-detector-MQTT2HA-Daemon #13: I2C Write Error

lightmaster, 2020-09-10:
**Checklist:**

- [X] I updated to the latest version available
- [X] I double-checked my hardware connections
- [X] I checked that my MQTT broker is otherwise working

**Release with the issue:**
v2.2.3
**Last working release (if known):**

**Hardware, Operating System, Python version:**
RPi 3B
Rasbian
Python 3.7.3

**Description of problem:**
Probably related to #10, but getting an I<sup>2</sup>C read/write error.

**Python errors shown in the logs (if applicable):**

```shell
pi@LightningPi [03:11:24 PM] [/opt/ISP-lightning-mqtt-daemon] [master *]                                                                                                      [9/190]
-> % python3 ./ISP-lightning-mqtt-daemon.py -d -v
[2020-09-10 15:11:32] - ISP-lightning-mqtt-daemon.py v2.2.3
[2020-09-10 15:11:32] - Verbose enabled
[2020-09-10 15:11:32] - (DBG): Debug enabled
[2020-09-10 15:11:32] - (DBG): * init mqtt_client_connected=[False]
[2020-09-10 15:11:32] * Sensor on I2C bus
[2020-09-10 15:11:32] - * Connecting to MQTT broker ...
[2020-09-10 15:11:32] - (DBG): log: Sending CONNECT (u1, p1, wr1, wq0, wf1, c1, k60) client_id=b''
[2020-09-10 15:11:32] - (DBG): log: Sending PUBLISH (d0, q0, r0, m1), 'b'home/nodes/sensor/lightningdetector/status'', ... (6 bytes)
[2020-09-10 15:11:32] - (DBG): * Wait on mqtt_client_connected=[False]
[2020-09-10 15:11:32] - (DBG): log: Received CONNACK (0, 0)
[2020-09-10 15:11:32] MQTT connection established
[2020-09-10 15:11:32]
[2020-09-10 15:11:32] - (DBG): on_connect() mqtt_client_connected=[True]
[2020-09-10 15:11:33] - (DBG): - stopped MQTT timer
[2020-09-10 15:11:33] - (DBG): - started MQTT timer - every 60 seconds
[2020-09-10 15:11:33] Announcing Lightning Detection device to MQTT broker for auto-discovery ...
[2020-09-10 15:11:33] - (DBG): log: Sending PUBLISH (d0, q1, r1, m2), 'b'homeassistant/sensor/lightningdetector/last/config'', ... (534 bytes)
[2020-09-10 15:11:33] - (DBG): log: Sending PUBLISH (d0, q1, r1, m3), 'b'homeassistant/sensor/lightningdetector/energy/config'', ... (301 bytes)
[2020-09-10 15:11:33] - (DBG): log: Sending PUBLISH (d0, q1, r1, m4), 'b'homeassistant/sensor/lightningdetector/distance/config'', ... (336 bytes)
[2020-09-10 15:11:33] - (DBG): log: Sending PUBLISH (d0, q1, r1, m5), 'b'homeassistant/sensor/lightningdetector/count/config'', ... (298 bytes)
[2020-09-10 15:11:33] - (DBG): log: Received PUBACK (Mid: 2)
[2020-09-10 15:11:33] - (DBG): log: Sending PUBLISH (d0, q1, r1, m6), 'b'homeassistant/sensor/lightningdetector/settings/config'', ... (418 bytes)
[2020-09-10 15:11:33] - (DBG): log: Received PUBACK (Mid: 3)
[2020-09-10 15:11:33] - (DBG): log: Received PUBACK (Mid: 4)
[2020-09-10 15:11:33] - (DBG): log: Sending PUBLISH (d0, q1, r1, m7), 'b'homeassistant/sensor/lightningdetector/crings/config'', ... (406 bytes)
[2020-09-10 15:11:33] - (DBG): log: Sending PUBLISH (d0, q1, r1, m8), 'b'homeassistant/sensor/lightningdetector/prings/config'', ... (403 bytes)
[2020-09-10 15:11:33] - (DBG): log: Received PUBACK (Mid: 5)
[2020-09-10 15:11:33] - (DBG): log: Received PUBACK (Mid: 6)
[2020-09-10 15:11:33] - * I2C configuration bus=1 - addr=3
[2020-09-10 15:11:33] - (DBG): - Testing AS3935 Communications...
[2020-09-10 15:11:33] - (DBG): - TEST write=5, read-back=5
Traceback (most recent call last):
  File "./ISP-lightning-mqtt-daemon.py", line 954, in <module>
    detector.set_noise_floor(testValue)
  File "/opt/ISP-lightning-mqtt-daemon/AS3935/AS3935_i2c_spi.py", line 194, in set_noise_floor
    self.write_byte(0x01, (self.read_byte(0x01) & 0b10001111) + ((noise_floor & 0x07) << 4))
  File "/opt/ISP-lightning-mqtt-daemon/AS3935/AS3935_i2c_spi.py", line 588, in write_byte
    self.pi.i2c_write_byte_data(self.device, address, value)
  File "/usr/lib/python3/dist-packages/pigpio.py", line 2743, in i2c_write_byte_data
    self.sl, _PI_CMD_I2CWB, handle, reg, 4, extents))
  File "/usr/lib/python3/dist-packages/pigpio.py", line 979, in _u2i
    raise error(error_text(v))
pigpio.error: 'I2C write failed'
[2020-09-10 15:11:33] - (DBG): log: Received PUBACK (Mid: 7)
[2020-09-10 15:11:33] - (DBG): log: Received PUBACK (Mid: 8)
```

**Additional information:**

ironsheep, 2020-09-10:
I see an addressing problem with the script. (reading/writing the wrong register) I'm chasing it!

ironsheep, 2020-09-10:
Sigh, nope not the wrong register - just new code resetting chip to factory defaults... and calibrating internal RC oscillators.  My bad.   Still hunting for possible issues.

ironsheep, 2020-09-10:
Couldn't find any cause for write errors.  Did find and fix a crash upon detect of overhead event.

Changes committed to /master.  Fixes appear in v2.2.5 and later.

ironsheep, 2020-09-10:
Please update to the latest version and then rerun with -d -v and let's see what we have...

lightmaster, 2020-09-10:
Whelp, looks like the I<sup>2</sup>C issue is likely nothing to do with the script, cuz this is the output of `sudo i2cdetect -y 1`

```shell
pi@LightningPi [06:41:20 PM] [~] 
-> % sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```

I'm at work for the night, so won't be able to double check the hardware until tomorrow. Not sure what could have failed, unless my breadboard is failing. I've another one I can swap with to test.

lightmaster, 2020-09-10:
Also, apologies for making you look through the code when its likely not the culprit. Haven't used I<sup>2</sup>C before and just found out about that command to test connection.

ironsheep, 2020-09-10:
Like minds, I was just going to ask you to check that!  Good work!
I'll be standing by to hear of your next testing...

You might want to double-check that SPI and I2C are both enabled via `sudo raspi-config` if you are going to be switching between, too.

ironsheep, 2020-09-10:
it's also a good idea to make sure your RPi has the latest and best of the software you are using. I keep mine up-to-date with

```shell
sudo apt-get update
sudo apt-get dist-upgrade
```

I answer yes to the list of updates so all are updated.

ironsheep, 2020-09-10:
Also, a couple more easy checks:

```shell
ls /dev/i2c*
# which yields something like (only if the driver is loaded):
crw-rw---- 1 root i2c 89, 1 Sep  1 13:17 /dev/i2c-1
```

or SPI:

```shell
ls /dev/spi*
# which yields something like (only if the driver is loaded):
crw-rw---- 1 root spi 153, 0 Aug 31 18:23 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Aug 31 18:23 /dev/spidev0.1
```

lightmaster, 2020-09-11:
Everything seemed good, but still nothing, so I ended up wiping the MicroSD card and starting from scratch. After updating everything, including using `sudo rpi-update`, it was detected using `sudo i2cdetect -y 1`. Must have been something unknowingly fubared.

closed 2020-09-11 by lightmaster