ironsheep/lightning-detector-MQTT2HA-Daemon #14: [Improvement] Hardlink to system tools

lightmaster, 2020-09-11:
https://github.com/ironsheep/lightning-detector-MQTT2HA-Daemon/blob/9415e4d7b0b74a7695e6599d151587d064e4f274/ISP-lightning-mqtt-daemon.py#L402

https://github.com/ironsheep/lightning-detector-MQTT2HA-Daemon/blob/9415e4d7b0b74a7695e6599d151587d064e4f274/ISP-lightning-mqtt-daemon.py#L407

I'd suggest looking in common system tool directories (`/bin`, `/sbin`, etc) and then hardlinking to system tools to make sure they work regardless of whether or not their directories are in the $PATH variable. Rasbian had a bug at one point where /sbin was not in the user's default $PATH and I just ran into this issue after installing zsh, with the script throwing an error that `ifconfig` was not found.

ironsheep, 2020-09-11:
Yep, this was my first ... I'll adopt the same mech I use in my other newer scripts. No use of the PATH variable!  Good observation, thanks!

ironsheep, 2020-09-11:
Changes committed to /master.  Fixes avail. in v2.2.6 and later.

closed 2020-09-13 by ironsheep