ironsheep/lightning-detector-MQTT2HA-Daemon #16: Provide an MQTT subscription triggered version

LouPaloma, 2020-10-22:
**The Lovelace component works great... but can't use it for my own sensor**
So I've created my own remote sensor wired to an ESP32 powered by a solar panel and LION battery.  It publishes strikes via MQTT.   For me, Python makes my head spin (yet I my ESP32 is in C++... ain't that a kick).

**How about making a version of this that is driven by an incoming MQTT subscription**
Can you yank out the SPI/I2C code and make this purely driven by an MQTT subscription?  I've found it quite easy to ETL my own published messages into your `detect` format based on an HA automation.  All that missing is a way to feed it back into your process.

While your at it, can you make this script something that HA can manage either in the `scripts/` or `python_scripts/` subfolders?

**Describe alternatives you've considered**
None, really... I like your Lovelace component and would love to add it...

**Additional context**
Here's a stupid simple template on reformatting my own message into your `detect `message:

```
- alias: Republish Lightning Message
  trigger:
    platform: mqtt
    topic: 'sensor/weather/test'
  action:
    service: mqtt.publish
    data_template:
      topic: 'sensor/weather/lightning/detect'
      #retain: true
      payload: >-
        {"last":{{ trigger.payload_json["sensor"]["timeStamp"] }},"energy":{{ trigger.payload_json["weather"]["energy"] }},"distance":{{ trigger.payload_json["weather"]["distanceMiles"] }},"count":1}
```

ironsheep, 2020-10-23:
@LouPaloma I'm not quite sure what you are asking for.  The card is already an MQTT listener.

Have you seen this? https://github.com/ironsheep/lightning-detector-MQTT2HA-Daemon/blob/master/THEOPS.md

If your sensor is reporting to MQTT you can adjust the content of what you are reporting per the document linked above and the existing card will then respond to your sensor.

This help?

LouPaloma, 2020-10-28:
Thanks.  I did look at THEOPS.  My point is that the daemon:

1) runs on a host that is physically connected to the sensor
2) all aggregations are done in daemon
3) The Lovelace card is "dumb" in that it gets ALL of it's display data for the presentation as part of the mqtt message for current or prior storms.

My sensor is remote, powered by a battery and solar panel such that it sends out minimal data packets, i.e., new strikes, via MQTT (or LoRa).

What I envision is your daemon would listen to an MQTT topic for a new strike (**_detect_** message format) and do all the aggregation that it currently does and publish the it's same data out onto topics.  

In short, I'm just looking to move the sensor input to come from an MQTT topic, but still be able to use your daemon and card.

ironsheep, 2020-10-29:
@LouPaloma Ok, now I understand.  However, if you take my daemon, make it listen to your sensor data coming in via MQTT it can then aggregate your incoming data and send the newly aggregated data to the lightning card as it does today.  Just rewrite the front end of my daemon to listen to your sensor.

Does this make sense?

LouPaloma, 2020-10-29:
Yes, it makes sense, but 1) I hate Python and 2) I have too many projects queued to take on another.  My assumption was that you could do what you mention in under 30 minutes as opposed to many hours of my "hacking".

The underlying thought is that you could increase usage of your front-end component (which is very nice [the daemon is very nice, too!]) by providing a more generic input method.  Per my initial issue comments, it'd be easy for anyone getting lightning data from any arbitrary source to re-format and re-post to your generic listener and then display it in HA.

I can understand if this isn't anything of interest, but thought you'd appreciate the kudos and input.

ironsheep, 2020-11-01:
@LouPaloma ok, thanks for the request and thank you for your kind words about this project I enjoyed creating it a I learned a lot in the process which is always our goal, right!

As you have already guessed I too have too numerous to count projects in the works... This is precisely why I wrote the THEOPS so I could enable others to extend as they wish while I keep working on more projects.   ;-)

Should you decide to take this on, feel free to make a pull request to get your work added to this project, should you wish. Just be sure to document the mechanism so others can make minor tweaks if their interface is not the same as what you constructed for yourself.

Again, Thank you!