ironsheep/lightning-detector-MQTT2HA-Daemon #17: Spi bus tuning_cap Error

keyj9, 2020-12-18:
Python error log:

pi@raspberrypi:~ $ python3 /opt/ISP-lightning-mqtt-daemon/ISP-lightning-mqtt-daemon.py --calc_tuning_cap
[2020-12-18 03:16:54] * Mode: Calculate Tuning Cap value and exit
[2020-12-18 03:16:54] - * init mqtt_client_connected=[False]
[2020-12-18 03:16:54] * Sensor on SPI bus
[2020-12-18 03:16:54] - - ip=[10.10.10.164], mac[d0:6f:4a:fe:64:f0], interface=[wlan0], uniq-id=[AS3935-d06f4afe64f0]
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
[2020-12-18 03:16:55] - - Testing AS3935 Communications...
[2020-12-18 03:16:55] - - TEST write=5, read-back=5
[2020-12-18 03:16:55] - - TEST write=2, read-back=2
* Please allow a long time for this function to stop. It should take a little over 3 minutes to test the 16 values
Traceback (most recent call last):
  File "/opt/ISP-lightning-mqtt-daemon/ISP-lightning-mqtt-daemon.py", line 1171, in <module>
    detector.calculate_tuning_cap()
  File "/opt/ISP-lightning-mqtt-daemon/AS3935/AS3935_i2c_spi.py", line 442, in calculate_tuning_cap
    freqs.append(self.calculate_resonance_frequency(seconds_try))
  File "/opt/ISP-lightning-mqtt-daemon/AS3935/AS3935_i2c_spi.py", line 463, in calculate_resonance_frequency
    cb = self.pi.callback(self.irq)
  File "/usr/lib/python3/dist-packages/pigpio.py", line 4920, in callback
    return _callback(self._notify, user_gpio, edge, func)
  File "/usr/lib/python3/dist-packages/pigpio.py", line 1210, in __init__
    self._notify.append(self.callb)
AttributeError: 'NoneType' object has no attribute 'append'

*I tried changing the intr_pin in the config.inf file but with no luck*

ironsheep, 2020-12-22:
Most of the buses on your RPi  (SPIx, I2Cx, etc.) need to be set up and or enabled.

Check to see if SPI is enabled with `ls /dev/spi*`

if you see something like:
`crw-rw---- 1 root spi 153, 0 Aug 31 18:23 /dev/spidev0.0`
`crw-rw---- 1 root spi 153, 1 Aug 31 18:23 /dev/spidev0.1`

then SPI0 is enabled.  if not use **raspi-config** to enable it.
if you are connected to SPIx which is not SPI0 then you will also need to enable a driver for it by editing /boot/config.txt

Also, it looks like you may not have started the **pigpiod service**.

Please follow the instructions to enable and then start the pgpiod.serice.
👍 
Also make sure your interrupt line is on the GPIO pin you expect and that gpio pin number is correctly entered into your config.ini

Does this help?

keyj9, 2021-01-14:
It turned out to be a hardware problem 
I tried another raspberry pi and everything worked fine

closed 2021-01-14 by keyj9