ironsheep/lightning-detector-MQTT2HA-Daemon #18: Sensor settings doesn't reflect what is configured in the config.ini

somejavadev, 2021-01-06:
**Checklist:**

- [x] I updated to the latest version available
- [x] I double-checked my hardware connections
- [x] I checked that my MQTT broker is otherwise working

**Release with the issue:**

`master` branch

**Last working release (if known):**

**Hardware, Operating System, Python version:**

Python: Python 3.7.3
OS: raspbian light
Hardware: RaspberryPi ZeroW

Following the manual installation via source code.

**Description of problem:**

Overall the integration is working but when I try and adjust the following properties in the config.ini file they don't seem to take effect.

```ini
detector_afr_gain_indoor = False

detector_noise_floor = 2
```

When the deamon starts the settings Data being logged via mqtt doesn't seem to match with the config I specified:

```
Publishing to MQTT topic "home/nodes/sensor/lightningdetector/settings, Data:{"settings": {"timestamp": "2021-01-06T19:59:49+00:00", "hardware": {"min_strikes": 5, "afe_inside": true, "disp_lco": false, "noise_floor": 1}, "script": {"period_minutes": 5, "end_minutes": 30, "number_rings": 5, "distance_units": "km"}}}"
```
The noise floor is explicitly logged as: `"noise_floor": 1` and not 2 and after a while you can see it is busy adjusting the noise_floor:

```
[2021-01-06 20:09:54] << INTR(17) >> Noise level too high - adjusting
[2021-01-06 20:10:23] << INTR(17) >> Noise level too high - adjusting
```

Ideally I want both those settings to take effect when starting the daemon.

**Python errors shown in the logs (if applicable):**

```

```

**Additional information:**

My config.ini, passwords redacted.

```ini
# Configuration file for lightning-detector-MQTT2HA-Daemon
# Source: https://github.com/ironsheep/lightning-detector-MQTT2HA-Daemon
#
# Uncomment and adapt all settings as needed.
# Some settings can be configured by environment variables.
# If an env variable is set, it takes precedence over settings in this file

[General]

[Daemon]

# Enable or Disable an endless execution loop (Default: true)
#enabled = true

[MQTT]

# The hostname or IP address of the MQTT broker to connect to (Default: localhost)
# Also read from the MQTT_HOSTNAME environment variable
hostname = xx

# The TCP port the MQTT broker is listening on (Default: 1883)
# Also read from the MQTT_PORT environment variable
port = 1883

# Maximum period in seconds between ping messages to the broker. (Default: 60)
#keepalive = 60

# NOTE: The MQTT topic used for this device is constructed as:
#  {base_topic}/{sensor_name}
#
# The MQTT base topic to publish the Lightning detector sensor data topics under.
base_topic = home/nodes

# The MQTT name for this Lightning detector sensor
sensor_name = lightningdetector

# The MQTT broker authentification credentials (Default: no authentication)
# Will also read from MQTT_USERNAME and MQTT_PASSWORD environment variables
username = xx
password = xx

# Enable TLS/SSL on the connection
#tls = false

# Path to CA Certificate file to verify host
#tls_ca_cert =

# Path to TLS client auth key file
#tls_keyfile =

# Path to TLS client auth certificate file
#tls_certfile =

# Broker connection-failure recovery
#  Rety connection attempts [default 5]
#retry_count = 5

#  Retry after waiting N seconds [default 30]
#retry_wait_in_seconds = 30

[Behavior]

# This script accumulates detections into buckets (rings if you will) for this period of time [2-10] in minutes [Default: 5]
#period_in_minutes = 5

# This script accumulates values into [3-7] rings [Default: 5]
#number_of_rings = 5

# This script can report distances in miles or kilometers (mi, km) [Default: km]
#distance_as = km

# This script determines that a storm has ended after this period of time [10-60] in minutes [Default: 30]
#end_storm_after_minutes = 30

[Sensor]

# decribe how your sensor is hooked up to your RPi
#  use values 'I2C' or 'SPI' - default is 'I2C'
sensor_attached = I2C

# GPIO pin used for interrupts
intr_pin = 17

# The SPI bus and device numbers come from the device file being used.
#   e.g., /dev/spidev0.0  would be bus=0, device=0
# The defaults are 0 and 0 (SPI0 which appears at the GPIO 40pin header)
#spi_bus = 0
#spi_device = 0

# Rev. 1 Raspberry Pis should leave bus set at 0, while rev. 2 Pis should set
# bus equal to 1. The address should be changed to match the address of the
# sensor.
#i2c_bus = 1
i2c_address = 0x03

# Value to use for your board
#   Internal Tuning Capacitors (from 0 to 120pF in steps of 8pf) - A value of [0-15]
# run the script with a --tune paramater to determine value for your board
#  NOTE: this runs for 3 minutes so be patient!  Then record your best value here.
#tuning_capacitor = 0x1

# Indoors (True) = more sensitive (can miss very strong lightnings)
# Outdoors (False) = less sensitive (can miss far away lightnings)
detector_afr_gain_indoor = False

detector_noise_floor = 2

# Prevent single isolated strikes from being logged => interrupts begin after
# this number of strikes (def: 5, value 1,5,9,16), then are fired normally.
#detector_min_strikes = 5

```

ironsheep, 2021-01-07:
I'm looking into this, more soon!

ocona040, 2022-06-09:
@ironsheep To follow up, I'm seeing this too. The Indoor/Outdoor param doesn't take effect as seen in the MQTT settings topic

jjskaife, 2024-05-05:
for the issue with indoor, the config parser statements need "config.getboolean"

https://stackoverflow.com/questions/12750778/booleans-in-configparser-always-return-true