ironsheep/lightning-detector-MQTT2HA-Daemon #19: Question on noise level

julianrinaldi, 2021-03-14:
**Checklist:**

- [x ] I updated to the latest version available
- [ x] I double-checked my hardware connections
- [x ] I checked that my MQTT broker is otherwise working

**Release with the issue:**
2.2.7
**Last working release (if known):**

**Hardware, Operating System, Python version:**
Raspberry Pi 3+, RaspiOS Debian Version 10.7, Python 3.7.3

**Description of problem:**

I'm getting a message saying 

```
[2021-03-14 12:39:13] << INTR(17) >> Noise level too high - adjusting
[2021-03-14 12:39:13] << INTR(17) >> Noise level too high - adjusting
[2021-03-14 12:39:14] << INTR(17) >> Disturber detected. Masking subsequent disturbers
```
I'm wondering what these messages mean, and if I should change a setting. Also, my tunning values seem pretty far "off" compared to your examples. The closest one is 0xf which is still +519.9. Is this ok, or is there something wrong?

**Python errors shown in the logs (if applicable):**

```

```

**Additional information:**

ironsheep, 2021-03-17:
Julian,

Please refer to the manufacturer datasheet for the detector chip. They have the most accurate explanation of the disturber settings.   Generally, I don't worry that the script is adjusting that value.

Regarding the calibration values grab a copy and paste them here.  I'll take a look.  It's been a while so I'll have to look things up after you provide what you are seeing.

julianrinaldi, 2021-04-21:
These are the values I'm getting:

```
pi@lightningdetector:/opt/ISP-lightning-mqtt-daemon $ python3 /opt/ISP-lightning-mqtt-daemon/ISP-lightning-mqtt-daemon.py --calc_tuning_cap
[2021-04-21 19:01:26] * Mode: Calculate Tuning Cap value and exit
[2021-04-21 19:01:26] - * init mqtt_client_connected=[False]
[2021-04-21 19:01:26] * Sensor on I2C bus
[2021-04-21 19:01:26] - - ip=[10.10.10.57], mac[b8:27:eb:d2:aa:00], interface=[wlan0], uniq-id=[AS3935-b827ebd2aa00]
[2021-04-21 19:01:27] - - Testing AS3935 Communications...
[2021-04-21 19:01:27] - - TEST write=5, read-back=5
[2021-04-21 19:01:27] - - TEST write=2, read-back=2
* Please allow a long time for this function to stop. It should take a little over 3 minutes to test the 16 values
For tuning 0x0: average frequency of 535924.852850 Hz (diff: +2245.3)
For tuning 0x1: average frequency of 535712.000818 Hz (diff: +2232.0)
For tuning 0x2: average frequency of 532472.834781 Hz (diff: +2029.6)
For tuning 0x3: average frequency of 534083.305766 Hz (diff: +2130.2)
For tuning 0x4: average frequency of 528674.481721 Hz (diff: +1792.2)
For tuning 0x5: average frequency of 526754.830675 Hz (diff: +1672.2)
For tuning 0x6: average frequency of 528261.016571 Hz (diff: +1766.3)
For tuning 0x7: average frequency of 524287.111727 Hz (diff: +1517.9)
For tuning 0x8: average frequency of 521586.103037 Hz (diff: +1349.1)
For tuning 0x9: average frequency of 520349.842987 Hz (diff: +1271.9)
For tuning 0xa: average frequency of 518043.465934 Hz (diff: +1127.7)
For tuning 0xb: average frequency of 517436.480007 Hz (diff: +1089.8)
For tuning 0xc: average frequency of 515508.947355 Hz (diff: +969.3)
For tuning 0xd: average frequency of 513032.312325 Hz (diff: +814.5)
For tuning 0xe: average frequency of 510722.559307 Hz (diff: +670.2)
For tuning 0xf: average frequency of 510976.786915 Hz (diff: +686.0)
- Your best tuning capacitor value is 0xe: which is off by +670.2
```

In Home Assistant the MQTT sensors all show as Unknown. Is this normal?

ironsheep, 2021-04-22:
It looks like your values are just fine.  The detector board, how your hardware is organized, and your electrical environment all play a part in altering the values determined by this script.  As long as the script finds a solution you should be good to go!

-Stephen

julianrinaldi, 2021-04-22:
Thanks for the response. Did you see my question about the MQTT sensors? They all show as Unknown. Is that normal? We had a thunderstorm yesterday and my Blitzortung integration showed lightning strikes close by, but the values on all of the sensors from this integration never changed. And I'm not actually sure which sensor to use with your lightning detector card. I'm assuming it's the sensor.lightningdetector_count and sensor.lightningdetector_last, but your directions for the card don't actually say which sensors to use.

I have your RPi reporter software and monitor card installed on about 20 raspberry pis around my home, so I just sent you a tip. I really appreciate your help with this.

ironsheep, 2021-05-04:
HI ok, if you are getting unknown the no data is being reported from your detector.

The python detector sense and report via MQTT script has a debug command parameter which you can enable to see whati's happening ... look for any problems when the script interacts with the detector.

Regarding which sensor to use I've added an example configuration that I use to the top-level README so you can see what I have that works.  See: https://github.com/ironsheep/lovelace-lightning-detector-card#card-configuration