ironsheep/lightning-detector-MQTT2HA-Daemon #20: Stupid Question from A New HA User

echellst, 2021-04-22:
Hello Stephen,

Thank you so much for this repo! I am a newly-converted Homekit to HA user and am simply in love with the fine work all of you have done.

i have a silly question, as I am also brand new to raspberry PIs/Linux. I am using Home Assistant OS 5.13, and it took me a while to figure out how to activate I2C, since the method you indicate was not possible (due to the Home Assistant OS). Here is the info I found on how to activate it (you may want to include this in your instructions) https://www.home-assistant.io/common-tasks/os/

My question is, how do I run the install script? I have no experience running an install script in Linux, and using  the terminal in HA OS, the sudo command fails. Could you please assist me here (or better yet, point me to where I can read how to do it?) I thrive on finding things out for myself, but in this case, I am stumped!

thank you so much for any help in advance.

Eric

ironsheep, 2021-04-22:
Hi Eric,

    The form you are using is actually a bit more difficult to do.  I use a separate RPi for HA (RPi 4 + large RAM) and then I use another RPi for my lighting detector.   This allows me to not have to worry about the limitations to command line access for the RPi running HA.  Oh, also there is a 2nd thing I did. I use the HA supervised install which actually does give me full access to the RPi 4 command line.  I did not use the HA OS form.

See: https://community.home-assistant.io/t/installing-home-assistant-supervised-on-a-raspberry-pi-with-debian-10/247116
While this page uses Debian 10, I actually just used the latest raspiOS build (which is derived from Debian.)   

You should be fine either way.

Once you have the lightning detector on a separate RPi my instructions should work well.

Does this help?

Regards,
Stephen

echellst, 2021-04-22:
It very much!
I think I’ll go that route. It sounds like a better plan than mine.

Thanks again for the information!

Eric

echellst, 2021-06-14:
Hi there.

About a month or so ago I reached out to you as I wanted to install your lightning detector on a PI4 with Home Assistant OS on it. You had advise to get another pi and just use the detector there, and MQTT to the Home Assistant one. I have done this (finally!) and have your card and software installed. But I am having a MQTT connection issue that I was wondering if you had a little time to assist me with. I have fully tested my Mosquito Broker, it works via internal, external (using Google MQTTLENS extension) and a few other devices that use MQTT. From that standpoint, everything seems to be working fine. But when I try to call python3 /opt/ISP-lightning-mqtt-daemon/ISP-lightning-mqtt-daemon.py, I get this response…
“ini mqtt_client_connected=[False]
MQTT connection error - broker @(MY IP) not responding. Retrying in….”  
And it tries 5 times and finnaly quits.

I have quadruple checked the host ip (my Home Assistant instance), port number, user and p/w. Everything matches perfectly. Any thoughts on what it could be?

Thank you so much for your time! 

Eric

ironsheep, 2021-06-14:
Sure...

Can you please run the script with debug output enabled? (ISP-lightning-mqtt-daemon.py -d -v <ret>)   It should fail the same way but should also provide more detail on what it's doing up to the failure.  Reply with that output and also include your config.ini in your reply as well.  Then let's see where we are once I see that...

echellst, 2021-06-14:
> Can you please run the script with debug output enabled? (ISP-lightning-mqtt-daemon.py -d -v ) It should fail the same way but should also provide more detail on what it's doing up to the failure. Reply with that output and also include your config.ini in your reply as well. Then let's see where we are once I see that...

Thanks so much for the help!

Here is the result…

***@***.***:~ $ python3 /opt/ISP-lightning-mqtt-daemon/ISP-lightning-mqtt-daemon.py -d -v
[2021-06-13 23:04:04] - ISP-lightning-mqtt-daemon.py v2.2.7
[2021-06-13 23:04:04] - Verbose enabled
[2021-06-13 23:04:04] - (DBG): Debug enabled
[2021-06-13 23:04:04] - (DBG): * init mqtt_client_connected=[False]
[2021-06-13 23:04:04] * Sensor on I2C bus
[2021-06-13 23:04:04] - * Connecting to MQTT broker ...
[2021-06-13 23:04:04] MQTT connection error - broker @(MY IP) not responding. Retrying in 30 seconds... (retry count = 5)
[2021-06-13 23:04:34] MQTT connection error - broker @(MY IP)  not responding. Retrying in 30 seconds... (retry count = 4)
[2021-06-13 23:05:04] MQTT connection error - broker @(MY IP) not responding. Retrying in 30 seconds... (retry count = 3)
[2021-06-13 23:05:34] MQTT connection error - broker @(MY IP) not responding. Retrying in 30 seconds... (retry count = 2)
[2021-06-13 23:06:04] MQTT connection error - broker @(MY IP) not responding. Retrying in 30 seconds... (retry count = 1)
^CTraceback (most recent call last):
  File "/opt/ISP-lightning-mqtt-daemon/ISP-lightning-mqtt-daemon.py", line 386, in <module>
    port=int(os.environ.get('MQTT_PORT', config['MQTT'].get('port', '1883'))),
ValueError: invalid literal for int() with base 10: '1883\n\n\n\n{base_topic}/{sensor_name}\nbase_topic = home/nodes'
 
During handling of the above exception, another exception occurred:
 
Traceback (most recent call last):
  File "/opt/ISP-lightning-mqtt-daemon/ISP-lightning-mqtt-daemon.py", line 393, in <module>
    sleep(mqtt_client_retry_delay_in_seconds) # some slack to establish the connection.

Eric Chellstorp

ironsheep, 2021-06-14:
Hmmm... Looks like there's something odd with your config.ini.

Have you specified a port number?  It should stay commented out if your MQTT broker is on the default port...

echellst, 2021-06-14:
Hmm, I bet that I have uncommented it. I’ll check a bit later and let you know! Thank you!

Eric Chellstorp

echellst, 2021-06-14:
Hi Stephen!

I did indeed uncomment the port. I  commented it out, re ran the test with debug and here is the result ( not I changed the attempts to only 2 and set the spacing to 15 seconds)

***@***.***:~ $ python3 /opt/ISP-lightning-mqtt-daemon/ISP-lightning-mqtt-daemon.py  -d -v
[2021-06-14 10:00:55] - ISP-lightning-mqtt-daemon.py v2.2.7
[2021-06-14 10:00:55] - Verbose enabled
[2021-06-14 10:00:55] - (DBG): Debug enabled
[2021-06-14 10:00:55] - (DBG): * init mqtt_client_connected=[False]
[2021-06-14 10:00:55] * Sensor on I2C bus
[2021-06-14 10:00:55] - * Connecting to MQTT broker ...
[2021-06-14 10:00:55] MQTT connection error - broker @192.168.4.41
 
{base_topic}/{sensor_name}
base_topic = home/nodes not responding. Retrying in 15 seconds... (retry count = 2)
[2021-06-14 10:01:10] MQTT connection error - broker @192.168.4.41
 
{base_topic}/{sensor_name}
base_topic = home/nodes not responding. Retrying in 15 seconds... (retry count = 1)
[2021-06-14 10:01:25] MQTT failed connect after 2 retries. Please check your settings in the configuration file "config.ini"

Eric Chellstorp